google/xls #2272: Can't use a parametric function in a calling function's parametric list

dank-openai, 2025-05-28:
**Describe the bug**

Motivating example: I want to use `apfloat::min_normal_exp<EXP_SZ>()` to define a new `apfloat:: min_subnormal_exp<...>`. My new function wants to return the smallest bitwidth that fits the integer returned. So naturally I call `apfloat::min_normal_exp` from within the parameter list to compute the returned bitwidth. But that's not working.

**To Reproduce**

Try to compile this:

```
// Returns the unbiased exponent of the smallest subnormal number representable by an apfloat
// with the given number of exponent and fraction bits.
pub fn min_subnormal_exp
    <EXP_SZ: u32, FRACTION_SZ: u32,
     R:
     u32 = {
         std::clog2(
             (-(apfloat::min_normal_exp<EXP_SZ>() as s32 - FRACTION_SZ as s32 - s32:1)) as u32) +
         u32:1} > () -> sN[R] {
     (apfloat::min_normal_exp<EXP_SZ>() as s32 - FRACTION_SZ as s32) as sN[R]
}

#[test]
fn test_min_subnormal_exp() {
    assert_eq(
        min_subnormal_exp<u32:8, u32:23>(),
        s9:-149);
}
```

and observe error:
```
1754:      u32 = {
1755:          std::clog2(
1756:              (-(apfloat::min_normal_exp<EXP_SZ>() as s32 - FRACTION_SZ as s32 - s32:1)) as u32) +
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^----^ TypeInferenceError: uN[32] Parametric expression `EXP_SZ` was not constexpr -- parametric values must be compile-time constants
1757:          u32:1} > () -> sN[R] {
```

**Expected behavior**
It should work without type error.

mikex-oss, 2025-05-28:
Are you sure it's because the called function is parametric? And not because it's imported? https://github.com/google/xls/issues/1941#issuecomment-2676662473

richmckeever, 2026-10-27:
This appears to succeed now.

closed 2026-10-27 by richmckeever