google/xls #2351: [enhancement][dslx] For a width slice with a const immediate start position >= size, give a warning

cdleary, 2025-06-06:
### What's hard to do? (limit 100 words)

Right now when imm is a u32 you can do imm[32 +: u32] and it will not flag that this is likely an error. This seems like a good warning to have in the DSLX warning set.

### Current best alternative workaround (limit 100 words)

Being cautious about this error pattern.

### Your view of the "best case XLS enhancement" (limit 100 words)

We give a warning for this case.

richmckeever, 2025-06-07:
This example is actually a hard failure in v2.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): Type System v2 added `WarningKind::kWidthSliceOutOfRange` (`xls/dslx/warning_kind.h:48`), emitted in `xls/dslx/type_system_v2/type_annotation_resolver.cc:996-1004` when a constexpr width-slice start or range exceeds the source bit width, and tested in `xls/dslx/type_system_v2/typecheck_module_v2_slice_test.cc:272-298`. (Closed by Gemini)

closed 2026-11-14 by allight