google/xls #2450: OSS build break: #embed doesn't work in CI

cdleary, 2025-06-22:
**Describe the bug**
```
Use --sandbox_debug to see verbose messages from the sandbox and retain the sandbox build root for debugging
bazel-out/k8-opt/bin/xls/build_rules/tests/arbitrary_data_file_embedded.cc:16:6: error: invalid preprocessing directive
   16 |     #embed "xls/build_rules/tests/some_arbitrary_file.txt"
      |      ^
bazel-out/k8-opt/bin/xls/build_rules/tests/arbitrary_data_file_embedded.cc:20:10: error: no matching function for call to 'MakeConstSpan'
   20 |   return absl::MakeConstSpan(data);
      |          ^~~~~~~~~~~~~~~~~~~
external/abseil-cpp~/absl/types/span.h:786:16: note: candidate template ignored: substitution failure [with ExplicitArgumentBarrier = <>, C = uint8_t[0]]: no matching function for call to 'MakeSpan'
  786 | constexpr auto MakeConstSpan(const C& c) noexcept -> decltype(MakeSpan(c)) {
      |                ^                                              ~~~~~~~~
external/abseil-cpp~/absl/types/span.h:791:[25](https://github.com/google/xls/actions/runs/15800697440/job/44538482747?pr=2449#step:8:26): note: candidate template ignored: substitution failure [with ExplicitArgumentBarrier = <>, T = uint8_t, N = 0]: zero-length arrays are not permitted in C++
  791 | constexpr Span<const T> MakeConstSpan(const T (&array)[N]) noexcept {
      |                         ^                              ~
external/abseil-cpp~/absl/types/span.h:773:25: note: candidate function template not viable: requires 2 arguments, but 1 was provided
  773 | constexpr Span<const T> MakeConstSpan(absl::Nullable<T*> ptr,
      |                         ^             ~~~~~~~~~~~~~~~~~~~~~~~
  774 |                                       size_t size) noexcept {
      |                                       ~~~~~~~~~~~
external/abseil-cpp~/absl/types/span.h:779:15: note: candidate function template not viable: requires 2 arguments, but 1 was provided
  779 | Span<const T> MakeConstSpan(absl::Nullable<T*> begin,
      |               ^             ~~~~~~~~~~~~~~~~~~~~~~~~~
  780 |                             absl::Nullable<T*> end) noexcept {
      |                             ~~~~~~~~~~~~~~~~~~~~~~
2 errors generated.
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 62.[34](https://github.com/google/xls/actions/runs/15800697440/job/44538482747?pr=2449#step:8:35)7s, Critical Path: 42.20s
INFO: 1990 processes: 1542 internal, 448 processwrapper-sandbox.
ERROR: Build did NOT complete successfully
```

https://github.com/google/xls/commit/f12d4877e9062153870596a3b6269a92a5bc5538#diff-9e80bd91d43dee3ec177e4665e564c0d5845a78863d89be19d1f7e4928ed04b2R16

OSS CI is broken because of this rule -- we should probably just do byte escapes since we're interpolating a template anyway?

cdleary, 2025-06-22:
cc @allight

ericastor, 2025-08-08:
I'm confused; I thought we'd configured things to only use `#embed` when it's supported by the compiler. Are people saying that this isn't true, and we're not sufficiently restricted? Or is this an argument that we shouldn't use `#embed` even when the compiler supports it?

hzeller, 2025-08-08:
Correct. It always uses `#embed` and [errors out](https://github.com/google/xls/blob/main/xls/dev_tools/embed_data/embedded_data.cc.tmpl#L10) if the compiler is not clang >= 19

hzeller, 2025-08-08:
ah ok, maybe I had `-Werror` on. If both things are handled, maybe don't even emit a warning if it would work both ways.

allight, 2025-08-08:
I mean I put that warning there mostly so we'd be able to see if it got broken instead of just silently falling back.

@cdleary is this still an active issue? If so can you tell me the compiler you're using so I can rework the if-def guards to support it. If it is just that it has a #warning I can also remove that I guess.