google/xls #2917: XLS crashes when converting code with fixed_point::make_fixed_point

lsrcz, 2025-08-21:
**Describe the bug**
XLS crashes when converting my DSLX code.

**To Reproduce**

```bash
root@machine:~/xls-v0.0.0-8501-gfc7c0e31f-linux-x64$ cat a.x
import fixed_point;

pub fn f(sum1: sN[24]) -> (fixed_point::FixedPoint<s32, u32:1, u32:23>) {
    fixed_point::make_fixed_point<s32:-23>(sum1);
}

root@machine:~/xls-v0.0.0-8501-gfc7c0e31f-linux-x64$ ./ir_converter_main a.x
E0821 04:46:15.325163 2022218 deduce.cc:1720] INTERNAL: XLS_RET_CHECK failure (xls/dslx/type_system/deduce.cc:1720) std::holds_alternative<Expr*>(eot)
0x5df1c06c2468: xabsl::StatusBuilder::CreateStatusAndConditionallyLog()
0x5df1c02dc0f9: absl::lts_20250512::StatusOr<>::StatusOr<>()
0x5df1c02eb379: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleTypeRefTypeAnnotation()
0x5df1c05c1f2b: xls::dslx::TypeRefTypeAnnotation::Accept()
0x5df1c02dba7f: xls::dslx::Deduce()
0x5df1c02d85e5: std::__1::__function::__func<>::operator()()
0x5df1c047020c: xls::dslx::DeduceCtx::Deduce()
0x5df1c02e9d5c: xls::dslx::(anonymous namespace)::DeduceVisitor::HandleTupleTypeAnnotation()
0x5df1c05c3dbb: xls::dslx::TupleTypeAnnotation::Accept()
0x5df1c02dba7f: xls::dslx::Deduce()
0x5df1c02d85e5: std::__1::__function::__func<>::operator()()
0x5df1c047020c: xls::dslx::DeduceCtx::Deduce()
0x5df1c047103c: xls::dslx::DeduceCtx::DeduceAndResolve()
0x5df1c02d8dbf: xls::dslx::TypecheckFunction()
0x5df1c02d4e7d: std::__1::__variant_detail::__visitation::__base::__dispatcher<>::__dispatch[abi:ne190107]<>()
0x5df1c02d364f: xls::dslx::typecheck_internal::TypecheckModuleMember()
0x5df1c02d3edc: xls::dslx::TypecheckModule()
0x5df1c02d2cdb: xls::dslx::TypecheckModule()
0x5df1c024a5ff: xls::dslx::ConvertFilesToPackage()
0x5df1c01f795f: main
0x7a8f558a01ca: [unknown]
0x7a8f558a028b: __libc_start_main
0x5df1c01f71e5: _start

E0821 04:46:15.325243 2022218 command_line_utils.cc:50] Could not extract a textual position from error message: INTERNAL: XLS_RET_CHECK failure (xls/dslx/type_system/deduce.cc:1720) std::holds_alternative<Expr*>(eot) : INVALID_ARGUMENT: Provided status is not in recognized error form: INTERNAL: XLS_RET_CHECK failure (xls/dslx/type_system/deduce.cc:1720) std::holds_alternative<Expr*>(eot)
Error: INTERNAL: XLS_RET_CHECK failure (xls/dslx/type_system/deduce.cc:1720) std::holds_alternative<Expr*>(eot)

root@machine:~/xls-v0.0.0-8501-gfc7c0e31f-linux-x64$ distro
Name: Ubuntu 24.04.1 LTS
Version: 24.04 (noble)
Codename: noble
```

**Expected behavior**
The compiler should not crash.

**Environment (this can be helpful for troubleshooting):**
Ubuntu 24.04

mikex-oss, 2025-08-21:
The error could definitely be better, but please note that there are multiple syntax issues with the code. The code should look something like:

```
import fixed_point;

pub fn f(sum1: sN[24]) -> fixed_point::FixedPoint<u32:24, u32:1, u32:23> {
    fixed_point::make_fixed_point<s32:-23>(sum1)
}
```

Namely:
1. Ending the body with `;` makes the function actually return the unit tuple `()`.
1. Wrapping the return type annotation with `(...)` makes it a tuple, so you would need to do the same in the body.
1. The actual issue flagged here (but poorly) is that the first parametric should be a `u32` value, not the type `s32`. See `pub struct FixedPoint<NUM_BITS: u32, EXPONENT_IS_NEGATIVE: u32, BINARY_UEXPONENT: u32>`.

allight, 2026-11-14:
Closing this issue as fixed (accepted by @allight): `xls/dslx/type_system_v2/inference_table_converter_impl.cc:378` now validates parametric argument kinds and returns a `TypeInferenceError` (`"Expected parametric value, saw `u32`"`) instead of failing a `RET_CHECK`, tested with the exact reproducer in `ValueTypeParametricMismatch` in `xls/dslx/type_system_v2/typecheck_module_v2_control_flow_test.cc:1289-1296`. (Closed by Gemini)

closed 2026-11-14 by allight