google/xls #2941: Migrate XLS to Bazel 8

mikex-oss, 2025-08-25:
Bazel 8 was released in December 2024. Notably, WORKSPACE is turned off by default, and language rules are no longer provided by core Bazel but loaded from repos like `@rules_cc` and `@rules_python`.

When upgrading the Python toolchain, I ran into an incompatibility with attempting to update `rules_python` to the latest release. Namely, when doing so, the `pip` repo rules attempt to generate `py_library`s with `pyi_srcs` and `experimental_venvs_site_packages` but using the native libraries...however, Bazel 7's built-in `py_library` doesn't have these attributes. I couldn't find a way to tell `pip` to use the language rules, so for now, I'm pinning `rules_python` to the version before these attributes were introduced (`1.0.0`).

I tried a straight update to Bazel 8 to support the newer `rules_python` version but ran into issues in `rules_hdl`'s Python pip setup. @mikesinouye mentioned we may not need the Python stuff in `rules_hdl`, so maybe that's fine. 

From https://github.com/google/xls/issues/931#issuecomment-2960052140, the `hier` project shows they are able to use rules_python 1.2.0 with Bazel 8, but they do not call  `init(...)` from `"@rules_hdl//:init.bzl`.

oharboe, 2025-09-04:
@QuantamHD @tspyrou @ollef @maliberty @povik FYI

I wanted to give Google XLS a go in https://github.com/The-OpenROAD-Project/RegFileStudy, but I don't have experience with using Bazel projects that use WORKSPACE.

Is there an ETA on fixing this?

In particular, I'm interested in whether Google XLS can break the back of variable latency pipelined operations such as division, floating point sqrt.

ORFS with yosys+abc retiming does a pretty good job on ASAP7 for 64x64 bit multiply, but for the floating point operations, the yosys+abc retiming is converges at 3x the clock period of the integer multiply case, which is a bit mysterious. Looks like yosys+abc doesn't do much on retiming after 3-4 pipeline stages, regardless of clock period.

[image]

[image]

oharboe, 2025-09-04:
> Can you use --enable_workspace ?

I prefer to wait until this is fixed and work other stuff meanwhile.

I had a quick look at the WORKSPACE file and the discussion in this PR and that made me hopeful that retiring WORKSPACE in this project isn't too far away.